davidmolony/MESC_Firmware #47: Request: enable sending log data via the term.

owhite, 2023-02-12:
Suggestion two levels of complicity for this. 

FIRST VERSION:

User select items for configuring the payload:

with a command like "status log init"

- frequency that payload is sent
- output format (e.g., "json", "csv")
- values to send, where the values correspond to what's available from get

And then when the user makes the call "status log send" the device starts sending the payload. 

SECOND VERSION:
If all of that is too complicated for now, much simpler, create a print statement that prints: 

`{"amps": 10, "volts": 20, "rpm": 30, "btemp": 40, "mtemp": 50}`

:-)

owhite, 2023-02-12:
Also: get should show temp data. :-)

davidmolony, 2023-02-12:
It would be very nice as well to have a second state that in the event of a crash/error state prints the values of the logged variables.
void  logVars(MESC_motor_typedef *_motor) implements this and is called from the hyperloop.
sampled_vars is the struct currently used for this as a ringbuffer; not used for both motors. Might want to make it configurable to sample either motor if we actually ever use two motors, but do not want two seperate buffers.

Printing the contents of this to the terminal/ESP32/whatever hopefully means that we can capture the events that led up to a trip/fire.