davidmolony/MESC_Firmware #63: Documentation for porting to other boards.

karu2003, 2023-11-04:
where you can read how to make an identification of the motor? There are GUI or CLI Tools for your project.
Is there a control of CAN BUS? DroneCan or so?  You write that MESC work to 100 kHz.
I am interested in this now. I have controllers working with the VESC. VESC does not work more than 45 kHz PWM.

davidmolony, 2023-11-04:
There's no documentation for porting to another controller. If you have f405 you can pretty much just copy one of the existing projects and modify the bits you see in the header like dividers and shunt values. The biggest issues are supporting different crystals to 8MHz and supporting hardware without BRK INPUT ON PB12. 

You can measure resistance inductance and flux linkage from the terminal by typing measure. I haven't invested much time in it because it's generally better to just use an inductance meter to get L and a current limited PSU/multimeter to get R.

Lambda is easy if you have an oscilloscope.

Yes it can run 100khz, by the way VESC counts it. However most hardware cannot do that frequency due to current measurement and gate driver limitations.

karu2003, 2023-11-05:
I have several EPC KITs to try.

Elump, 2024-08-26:
If you measure with RLC, you get phase to phase values since you are measuring on two wires of the motor. A division by 2 is needed to get the right value per phase.