davidmolony/MESC_Firmware #64: Its posible to create step/dir interface also??

artayasa, 2024-01-15:
Hi sir. 
It posible to add strp/dir interface also? Soa will posible to used bldc on cnc. 
Iam plan to used stm32f401. 
An make mybe 70v20A or bigger volt.

davidmolony, 2024-03-17:
Hi,
This is possible if running in open loop and perhaps position control mode with encoder.
Would need to have a timer input set up to count pulses, perhaps timer4 could be repurposed for this, and another pin which can be interrupted to change the count direction  of the timer4 with high priority.

artayasa, 2024-03-17:
> Hi, This is possible if running in open loop and perhaps position control mode with encoder. Would need to have a timer input set up to count pulses, perhaps timer4 could be repurposed for this, and another pin which can be interrupted to change the count direction of the timer4 with high priority.

Yes sir, but i still confiusd about method how to put the step/dir interface, now i have stm32 g431 with 48pin. 170Mhz. Tim1 for motor, and also suport incremental encorder on some pin. I learn already. But still confiusd when put the methode about it. And also for curent sensor if i make for like 70-200v, 5-8A max. Need change the curent sensor sir???

davidmolony, 2024-03-17:
This codebase is not currently ported to G431 so you have an extra struggle there. It is perfectly possible though.

Current sensors can be used on low side for that voltage, it is not really a property of the firmware.

The input timer setup would look something like this, and you would need to modify the Status register DIR bit with the direction pin toggle.

STM32 is not naturally an ideal choice for this due to the possibility of losing a pulse as the DIR toggles. This is generally an issue with MCUs, and is why industrially quadrature signals are preferred.
[image]

artayasa, 2024-03-17:
> industrially quadrature signals are preferred.

What do u mean of this sir?? It need quadature encorder? Because on my pmsm motor already have quadature encorder 2000ppr sir