davidmolony/MESC_Firmware #74: Alternative Field Weakening Method / control strategy

phkahler, 2025-05-22:
David,

Some years ago I patented this method, which we can discuss if you'd like to try it.

[This here expired patent](https://patents.google.com/patent/US8593091B2/en?oq=US+8.593%2c091+B2)

The patent has since expired due to non-maintenance.  I'm not sure how clear the text is, and I"m not sure where to get a proper .pdf of it these days.

I can tell you that in an IPM at maximum physically achievable torque, the required voltage angle can exceed 90 degrees from the BEMF / no load condition. I noticed how you implemented voltage limits in your DQ control, and figuring out how to do that optimally was the origin of the idea for the patent. I've seen various attempts to deal with voltage limits, including keeping the reference DQ current inside the limit to maintain some control authority (trading stability for efficiency in field weakening?). This algorithm operates best when maxed out hard against the voltage limit. AFAICT it is still mostly unknown.

phkahler, 2025-05-30:
I spent a great deal of time figuring out how to switch to polar reference frame and back. The answer was to implement it in the existing cartesian DQ coordinates (linearized around the current integrator vector). The polar controller can operate below voltage limit too, it just can go down to very low speeds. The gains for the conventional controller (or the error terms) are ramped down while the polar ones are ramped up. It ends up looking like some additional control code (extra terms) mixed in with the existing PI controllers.

What hardware do you recommend to run this repo on?

davidmolony, 2025-05-30:
Most VESC compatible hardware can run this, provided it is the newer 3 shunt types and the shunts are measured on PC0,1,2 and phase voltages on PA0,1,2 and Vbat measured on PC3

What is the voltage lead beyond 90 degrees you see in an extreme case? Just by considering Vmag = Vd^2+Vq^2, I realised I could allow Vd to get far closer to Vmag while still allowing considerable Vq... e.g. sin(10) = 0.17 and cos(10) = 0.985... so for a 1.5% drop on   Vd you get 10 degrees lead angle beyond 90, which I think is probably barely noticeable and a huge change over the current 60 degrees lead from the d axis...

Your method I think only becomes useful if you need to be running at advance angles much beyond 100degrees. If that is a realistic use case, then I should reconsider the overall method.