davidmolony/MESC_Firmware #75: Header file for FOC_KING boards

bmentink, 2025-05-26:
I noticed a video out there where MESC was used on the FOC_KING hardware.
Just wondering if anyone has the header file for that target, if not, will have a go myself porting to that board.

davidmolony, 2025-05-26:
You would have to contact the creator of the board, Jens Overby. I've seen the video but never had the hardware so never added it myself. 

Feel free to get/create it and send a pull request to add it to the list of boards under the name "freedrive".

bmentink, 2025-05-26:
Thanks for the prompt reply. I have had a quick took and started with duplicating the MP2  header, and making mods.   But I am struggling if there  is any other files that need modification, for example, the header file does not specify what pins are assigned to PWM, current sensors, voltage etc ... is there a guide for porting?
I will try and contact Jens Overby but have not found a link yet ..  Cheers..

bmentink, 2025-05-26:
Seems from the foc_king schematic that the voltage and current sens use different ADC channels, I have attached cpu diagram ..
I guess changes will need to be made in MESChw_setup.c ? [https://drive.google.com/file/d/16FIt_TGT1u8178ic4kibBDphKwrzrpmw/view?usp=sharing](url)

closed 2025-06-10 by davidmolony