davidmolony/MESC_Firmware #78: Difference between  par_motor_sensor and par_SL_sensor?

bmentink, 2025-06-07:
Hi,

I finally managed to get  MESC working on foc_king hardware.
It's mostly working, reads Vbus correctly and offset raw values around 2000 correct.
I did find that the motor params that I set up (compiled in) did not propagate to the values when on TTerm, I had to plug them in manually, and then the motor would turn.

Must say impressed, no judder start like VESC, very smooth starting.
However, I am confused with the above sensor modes. I don't know what mode I am running in. I have set par_SL_sensor=3(hfi) and  par_motor_sensor=0 (SL) Which mode am I in? WHat is the difference between these two.  Anything other than 0 for par_motor_sensor does not work.

I have looked at all the videos, but all the names in TTerm have changed with the latest version of your code. Will you be updating some docs any time soon?

Also, I can't seem to get "measure -r" and measure -f" to pick up the motor, return rubbish values every time. Even though I put in sensible values manually and it runs ok.

I also have not got the temp sensor on board (MOS)  to read correct temperature. Is there a calibrate function somewhere?

Thanks, keep up the good work.

davidmolony, 2025-06-07:
There will be limited updates to this firmware, since I'm very much taken up with other things, including commercial versions of this ...

I haven't used, tested or updated the measure functions for years. 

Regarding the SL and SL sensor, par_motor_sensor is the full range sensor getting used, the SL_sensor is like a startup sensor ... It runs only at low speed before switching over to the normal observer. 

Obviously for the other sensors you need to actually install them (encoder, hall etc).

bmentink, 2025-06-07:
Totally understand. Thanks for the time. Before I stop bugging you, could you please answer the following.

1.  With opt_cont_type torque and duty work fine with sending uart_req, but speed control option does not. Does it use the speed_req term? If so what are the units, as I put in a sensible erpm but it just faults with overcurrent.
2. What is uart_dreq?
3.  What are the meas_* terms, what do they do?
4. 
Many thanks, I will try and get the rest of my questioned answered from the code..

davidmolony, 2025-06-08:
Speed requires that you tune the PI parameters for it to work. It also requires that it can start up well, because otherwise the speed PI is getting a lot of noise at low speed.

uart_dreq is used for aligning the rotor or just testing efficiency by adding extra d axis current.

meas_ is for the measurement functions, you can define the amount of current/voltage it uses... but you are best off knowing the parameters a-priori.

beta value is defined in the stm32fxxx.h file for each project in the inc. Probably ought to be surrounded by #ifndef so that you can redefine it in your hardware conf...

bmentink, 2025-06-09:
Thanks David. One thing I forgot to ask you. How can I set the input parameters so that I have center 0 on for example, adc1 ..
(I want FWD/REV on the one throttle)
From the code it looks like if I set `adc1_min` to -2000 (4000 full throttle) then the calculations would give me the correct  negative torque, on the lower part of the throttle. However, in TTerm I cannot put in a negative value. Where can I change that ..

bmentink, 2025-06-10:
Never mind, this works:
```
void Boat_app(MESC_motor_typedef *_motor){
	static float temp;
	 _motor->safe_start[0] = 0;	// Remove deadband so we can apply our own.
	  // Make our pot center zero
	 temp = _motor->input_vars.ADC1_req  - 0.5;
	 // Deadband
	 if ((temp < 0.05) && (temp > -0.05)) {
		 temp = 0.0f;
	 }
	 //We just scale and sum the input current requests
	  _motor->FOC.Idq_prereq.q = 	_motor->input_vars.UART_req +
	  	  	  	  	  	  	  	  	_motor->input_vars.max_request_Idq.q * (temp + _motor->input_vars.ADC2_req +
	  	  	  	  	  	  	  	  	_motor->input_vars.RCPWM_req + _motor->input_vars.ADC12_diff_req +
									_motor->input_vars.remote_ADC1_req + _motor->input_vars.remote_ADC2_req );

	  //Clamp the Q component; d component is not directly requested
	  _motor->FOC.Idq_prereq.q = clamp(_motor->FOC.Idq_prereq.q, _motor->input_vars.min_request_Idq.q, _motor->input_vars.max_request_Idq.q);
	  //_motor->FOC.Idq_prereq.q += -1.0f;
}
```